SystemRDL/PeakRDL-regblock #157: [FEATURE] Add OBI protocol as default to other interfaces

DanielKellerM, 2025-08-15:
- [x] I have reviewed this project's [contribution guidelines](https://github.com/SystemRDL/PeakRDL-regblock/blob/main/CONTRIBUTING.md)

**Problem / Limitation**  
`PeakRDL-regblock` currently supports defining CPU interfaces such as APB and AXI-lite, but there is no built-in option for the [Open Bus Interface (OBI)](https://github.com/pulp-platform/obi).  
OBI is open-source, lightweight, and widely used in open hardware projects, so it would be a natural addition for users targeting that ecosystem.

**Proposed Solution**  
Include OBI as a built-in `Cpuif` option in `PeakRDL-regblock`.  
An initial implementation is available here for reference: [feat/obi branch](https://github.com/DanielKellerM/PeakRDL-regblock/tree/feat/obi).

**Alternatives Considered**  
- Use another protocol (e.g., APB, AXI-lite) and wrap it externally — but this adds latency and complexity compared to native OBI generation.

**Additional Context**  
I have verified the implementation here https://github.com/[DanielKellerM/peakrdl-obi-apb-verification](https://github.com/DanielKellerM/peakrdl-obi-apb-verification). Includes a self-checking SystemVerilog testbench that exercises reads, writes, byte-enable handling, and `RREADY` back-pressure. The tests pass on Questa 2023.4, and a minimal reproducible repo can be provided if helpful. The test includes comparison with an APB interface also generated by `PeakRDL-regblock`

FYI @micprog

micprog, 2025-08-15:
For context, the OBI specification can be found here: https://github.com/openhwgroup/obi/blob/072d9173c1f2d79471d6f2a10eae59ee387d4c6f/OBI-v1.6.0.pdf

amykyta3, 2025-09-19:
Agreed! Closing and will continue discussion in the PR you provided.

amykyta3, 2025-10-25:
Done. Will be in the next release

closed 2025-10-25 by amykyta3